MicrosoftDocs/windows-iotcore-docs #675: Raspberry PI 3 Bluetooth driver problem

Parsakarami, 2018-10-04:
Is the Raspberry PI 3 Bluetooth driver problem solved on October 2018 version of windows?

In the previous version of windows IoT core on raspberry pi 3 (model B) the internal bluetooth adapter works for 20 to 30 minutes perfectly but after this period of time it completely will be out of access by others as advertiser of a ble service, an no phone can find it.

Is the problem solved?

saraclay, 2018-10-09:
Hi @Parsakarami, our team is taking a look. Thanks!

saraclay, 2018-10-11:
Hi @Parsakarami, so it seems that there have been a few others that have reported that the on-board Bluetooth driver on the Raspberry Pi board isn't reliable when handling a high rate of sensor updates or are long-time running. Because the hardware does not implement compatible firmware or hardware flow control, the speed is limited to ~115kbps. High speed BT is not recommended on RPi/IoT Core using the onboard on RPi3 BT hardware. Does this help?

Parsakarami, 2018-10-13:
Hi and thanks @saraclay for your help, so is Microsoft going to resolve this issue? Or not? 
If yes does Microsoft has any ETA for this? 
If no does Microsoft resolve it in Raspberry Pi 3 B+ ?

Before answering my questions, thank you for the help you made.

saraclay, 2018-10-16:
Hi @Parsakarami, this is unfortunately a known issue on the Raspberry Pi itself. So we do not recommend high speed bluetooth and this does not resolve for B+.

closed 2018-10-16 by saraclay