MicrosoftDocs/windows-iotcore-docs #698: Pi3 suuport

Eg-Virus, 2018-10-07:
Hi, i tried this setup with Pi3 and built-in WiFi, but it didn't work out, can you help out in this?

Thank you

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7e1ef6b3-6d6e-465e-a542-6d6aeddd4fea
* Version Independent ID: c49e4c94-5bfb-43a3-42e6-12e00f299a3d
* Content: [Using WiFi Direct on your Windows 10 Iot Core Device - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/setupwifidirect)
* Content Source: [windows-iotcore/connect-your-device/SetupWifiDirect.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/SetupWifiDirect.md)
* Product: **windows-iot**
* GitHub Login: @derekameer
* Microsoft Alias: **demeer**

arnvanhoutte, 2018-10-09:
Doesn't work either for me. When I use my PC as advertiser I get an error "Connection request from PI was declined",

the other way around I get an error "A socket operation was attempted to an unreachable host"

Tamer-Gamal, 2018-10-09:
> 
> 
> I managed to fix the issue on my Pi 3B+ by adding Internet Server & Client capabilities in my appmanifest

Thank you

closed 2018-10-09 by Tamer-Gamal