MicrosoftDocs/windows-iotcore-docs #716: evaluation copy of Windows 10 Enterprise download link redirected

luyaok, 2018-10-11:
I use the "evaluation copy of Windows 10 Enterprise" to download enterprise evaluation image but failed, this link seems redirected.
https://www.microsoft.com/en-us/evalcenter/evaluate-windows-10-enterprise

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a1e7dc7e-3df6-2eaf-9c11-ed03c00719e0
* Version Independent ID: ab5e43cf-c4db-cd8f-e92a-c33e4ebc2b68
* Content: [Overview of Windows 10 IoT Enterprise - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/windows-iot-enterprise)
* Content Source: [windows-iotcore/windows-iot-enterprise.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/windows-iot-enterprise.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-10-16:
HI @luyaok, we are looking into this.

saraclay, 2018-10-17:
Hello again - the evaluation center would be putting back up the RS4 and 2016 LTSB bits soon (latest tomorrow). To learn more about the issue, you can read the blog here: https://blogs.windows.com/windowsexperience/2018/10/09/updated-version-of-windows-10-october-2018-update-released-to-windows-insiders/#KyvjHp6XBmGFdVrd.97 

If you find that you still cannot get any bits after tomorrow, feel free to open a new issue.

closed 2018-10-17 by saraclay