MicrosoftDocs/windows-iotcore-docs #720: Secure Boot Test Package not working

forceshutdown, 2018-10-14:
Hey Guys

I am trying to implement SecureBoot for my DragonBoard 410c. I am following this Guide and currently I have a Problem just with a test Build
https://docs.microsoft.com/en-us/windows/iot-core/secure-your-device/securebootandbitlocker

I am just using the Test Certificates and no additional Config to the XML. The Packet builds and I can flush it with the FFU. Windows boots up and enables Secure Boot. (It ist enabled in the UEFI Menu of the Dragonboard) but as soon as I restart the device, it does no longer boot windows.

I enabled RPMB and cleared the Keys before Flashing. Are there additional steps for a test deployment? I know that I need my own OEM Keys for Retail but I want to run test first.
[QCDB_settings.txt](https://github.com/MicrosoftDocs/windows-iotcore-docs/files/2476530/QCDB_settings.txt)

saraclay, 2018-10-16:
Hi @forceshutdown, we are looking into this.

imingc, 2018-10-17:
@forceshutdown - can you make sure that your BSP CABs are **resigned** with your code signing cert that is purchased from a CA and chained to cross signed root ?
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/build-retail-image

saraclay, 2018-10-22:
Hi @forceshutdown, 

According to this MSDN post:
https://msdn.microsoft.com/en-us/windows/compatibility/secured-boot-signing-requirements-for-kernel-mode-drivers

Test signing is disabled when secure boot is "on" - and test signing is needed to allow test certs.

closed 2018-10-22 by saraclay