MicrosoftDocs/windows-iotcore-docs #721: Spelling Error

ntindle, 2018-10-15:
&gt;If you haven't connected and display and would like to connect via Wi-Fi, you'll need to:
and -&gt; an

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6547b7e1-4e53-9510-e550-ac2d369da6d9
* Version Independent ID: a3c18768-7ee4-4e7a-2ca4-f6f6c31f342a
* Content: [Setting up your device - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/quickstarter/devicesetup#using-the-iot-dashboard-dragonboard-410c)
* Content Source: [windows-iotcore/tutorials/quickstarter/DeviceSetup.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/tutorials/quickstarter/DeviceSetup.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

ntindle, 2018-10-15:
and-> a

saraclay, 2018-10-16:
Thank you! Feel free to open a pull request next time :)

closed 2018-10-16 by saraclay