MicrosoftDocs/windows-iotcore-docs #726: Windows 10 IoT Core Packages for October 2018 update

Vake93, 2018-10-16:
Where can I download the Windows 10 IoT Core Packages for October 2018 update?, currently I only could find Packages for April update at https://www.microsoft.com/en-us/software-download/windows10IoTCore .

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e550244c-f4be-049b-9131-ba32bd3e9214
* Version Independent ID: 77ea5cc8-8a98-53e4-affd-a1f30538bee6
* Content: [October 2018 Update - Build 17763 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/currentcommercial)
* Content Source: [windows-iotcore/release-notes/CurrentCommercial.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/CurrentCommercial.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-10-16:
Hi @Vake93, we are looking into this.

saraclay, 2018-10-16:
Hi @Vake93, due to an issue with the Desktop OS, we are holding this update and validating new packages internally.

closed 2018-10-16 by saraclay

endyb66, 2018-10-16:
Hi @saraclay, is this also the reason, that currently now OTA update is comming for the october update?

Vake93, 2018-10-17:
@saraclay got it, thanks. Any Idea about a date for the release of the packagers?