MicrosoftDocs/windows-iotcore-docs #727: Dragonboard 410c analog MIC input

endyb66, 2018-10-16:
On my Dragonboard, i want to record some audio with an analog microphone. According to the notes, the drivers should be included for the analog audio header. I connected a headset as follows to the J7 heade on the Dragonboard:

Sleeve -&gt; J7 (MIC2_P)
Ring2 -&gt; J7 (AGND)
Ring1 -&gt; J7 (HPH_R)
Tip -&gt; J7 (HPH_L)

Also Looking at the available devices with PowerShell on my Dragonboard, i get the following output:

IoTCoreAudioControlTool l
d,r,Speakers (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.0.00000000}.{2ff88fd6-f868-4e21-b4f3-7242c6e16692},100.0

d,c,Internal Microphone (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.1.00000000}.{2fa3b5ac-187e-4cc8-a0d4-451a1b30b50b},100.0
,c,Internal Handset Microphone (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.1.00000000}.{31fcf5da-b92e-4615-8ba8-c8ee4f3bea36},100.0

Tested with both audio capture devices (Internal Microphone and Internal Handset Microphone), with both the same problem, all i hear is some noise on my recorded file. The microphone i'm using is tested and is working fine (e.g. on a Smartphone). 

So my question is what i'm doing wrong? Is there a Problem with my code only, or is there an issue with the audio capture device itself? Or is it the connection between the 3.5mm plug and the Dragonboard?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e550244c-f4be-049b-9131-ba32bd3e9214
* Version Independent ID: 77ea5cc8-8a98-53e4-affd-a1f30538bee6
* Content: [October 2018 Update - Build 17763 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/currentcommercial)
* Content Source: [windows-iotcore/release-notes/CurrentCommercial.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/CurrentCommercial.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-10-16:
We are looking into this.

moyanming, 2018-10-20:
Hi @endyb66 ,
There is a mic on the dragonboard 410c : GM1, this mic is connect to the MIC1 of the audio codec.
Could you please test this GM1? Make sure that you should set the default capture device to GM1.
Best regards.

moyanming, 2018-10-20:
Hi @endyb66 ,
I have made my custom pcb with MIC1 includes. I will feedback here after test.
My dragonboard 410c not installed GM1 too.

endyb66, 2018-10-20:
Ok thank you @moyanming.

saraclay, 2018-10-24:
Hi @endyb66 and @moyanming, please take a look at this guide from Qualcomm: file:///C:/Users/saclayt/AppData/Local/Packages/Microsoft.MicrosoftEdge_8wekyb3d8bbwe/TempState/Downloads/lm80-p0436-43_c_stereo_connector_and_audio_routing_on_dragonboard_410c_application_note%20(1).pdf

saraclay, 2018-10-24:
Hi @endyb66 and @moyanming, we're advising people to take a look at this guide from Qualcomm: https://developer.qualcomm.com/download/db410c/stereo-connector-and-audio-routing-application-note.pdf 

I'm closing this out for now. Feel free to reopen if the guide does not answer your questions.

closed 2018-10-24 by saraclay

endyb66, 2018-10-24:
Hi @saraclay, please reopen, this guide does not cover the problematic, which seems to be an issue with Windows 10 IoT Core.

IoTJonL, 2018-11-01:
I confirm the following setup to be working today:
1. 410c device running October Update release (17763).
2. Analog microphone connected to Mic1 connector
3. Using Internal Microphone driver
4. Using https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/SpeechRecognitionAndSynthesis

I noticed that you did not mention connecting J7 HPH_REF to Ring 2, which I think is required according to page 7 of the Qualcomm doc. Also there is a note on page 6 that mentions Sleeve and Ring 2 might be swapped on some jacks, so I would double check that too.

endyb66, 2018-11-02:
Thank you @IoTJonL, so an analog mic does work without issues on your side?

moyanming, 2018-11-05:
> I confirm the following setup to be working today:
>
> 1. 410c device running October Update release (17763).
> 2. Analog microphone connected to Mic1 connector
> 3. Using Internal Microphone driver
> 4. [...]

Hi @IoTJonL ,
The MIC1 (CDC_MIC1_P, GM1) which is on the Dragonboard 410c board is work fine. But this is not answer to the question. The analog Mic (J7) input which is the MIC2 (CDC_MIC2_P) can't work.
[image]
[image]