MicrosoftDocs/windows-iotcore-docs #743: Stuck at Rainbow Screen

italiann, 2018-10-17:
I am using Raspberry Pi 3+ and have installed windows 10 iot. I run Iot Dashboard to make a bootable SD card  but it failed. Then I made custom install with flash.ffu and it failed again. Failed means it stucks at the begining of the boot screen. Windows 10 iot mainpage or destop whatever you called it, never comes up. My SD card is bootable media, also I have tried with many SD cards. The result is the same. My power adaptor supports 2.5 Amphere and it works with raspberrian OS. If you assist me regarding subject I wil be appreciative.

saraclay, 2018-10-23:
Hi @italiann, which SD cards have you used? Could you list them? Strange that the main page does not come up as it seems that you at least make it to the boot screen.

izzis, 2018-10-24:
So do I.
I can only use RaspberryPi 3B+ Technical Preview Build 17661.

saraclay, 2018-10-24:
Hi @izzis - I see. I didn't realize you were using the 3B+.

Please take a look at this forum here to see if anything will help:
https://social.msdn.microsoft.com/Forums/en-US/64f33778-4c86-4460-9e4e-53348c73733e/raspberry-pi-3-b-windows-iot-core-boot-work-around?forum=WindowsIoT 

Depending on what you're looking to do with the 3B+, we are recommending that users stick with the 3B as it's just a technical preview.

closed 2018-10-25 by saraclay