MicrosoftDocs/windows-iotcore-docs #744: No Headset Earphone render device!

moyanming, 2018-10-18:
DragonBoard 410c, Audio Messanine Board, test on both Windows 10 IoT Core v10.0.17134.1 and v10.0.17763.1 OS.

I want to use the Headset Earphone and Headset Microphone on the Audio Mezzanine Board, the Headset Earphone have no sound, but the speaker have. It seems that there is no Headset Earphone render device. How to fix this?

Best regards.

Command> iotcoreaudiocontroltool l

d,r,Speakers (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.0.00000000}.{04136b70-4ad5-4528-940b-9c70421c5c38},100.0

,c,Internal Microphone (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.1.00000000}.{06720b65-3e95-48dd-9b94-51147ef3cfee},100.0
d,c,Internal Handset Microphone (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.1.00000000}.{4ffda199-4581-46f4-b27d-6c8fbae2e793},100.0

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f2599338-019c-10e6-13e6-620630f42b95
* Version Independent ID: caafa3ff-dfe6-5af5-efe2-978c3469c2b2
* Content: [Release Notes for Build 17744 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/17744)
* Content Source: [windows-iotcore/release-notes/Insider/17744.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/Insider/17744.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

moyanming, 2018-10-18:
[image]

endyb66, 2018-10-18:
Same Problem here with microphone

IoTCoreAudioControlTool l
 d,r,Speakers (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.0.00000000}.{2ff88fd6-f868-4e21-b4f3-7242c6e16692},100.0

d,c,Internal Microphone (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.1.00000000}.{2fa3b5ac-187e-4cc8-a0d4-451a1b30b50b},100.0
 ,c,Internal Handset Microphone (QCAUD Real Time (WaveRT) Driver for Qualcomm SoC),{0.0.1.00000000}.{31fcf5da-b92e-4615-8ba8-c8ee4f3bea36},100.0

I think it is related to my own issue here https://github.com/MicrosoftDocs/windows-iotcore-docs/issues/727

moyanming, 2018-10-18:
Hi @endyb66 ,
I have not test Headset Microphone yet, but why this can't be used as there is a "Internal Handset Microphone" device?
Best regard.

endyb66, 2018-10-18:
Hello @moyanming, i have no mezzaine board attached, i'm using the J7 connector directly. But none of both mics do work (when i set them as Default render device from PowerShell). Can you test, if it does work with your mezzaine board? I also need the line output, but this does also not work. only Thing that does work is to connect a speaker directly to the board (the two first Pins on J7).

Best Regards

moyanming, 2018-10-19:
Hi @endyb66 ,
The Audio Mazzaine Board is just connect the Audio Expansion directly from the Dragonboard.
I have test the Speaker (SPKR_OUT_P and SPKR_OUT_M) successfully. But the Headset Earphone (CDC_HPH_L and CDC_HPH_R) have no sound, also, there is no Headset Earphone render device at all.
I have not test the Microphone, nether Headset Microphone (CDC_MIC2_P), no the Microphone (CDC_MIC3_P) which is on the Audio Mazzaine Board "M1" (M2 is a digital Mic, I have not test M2). I don't know how to the the Microphone as I have no Audio test app. Do you have any idea or could you please give me an App just for the Audio test?
Best regards.
[image]
[image]
[productspecification-audiomezzanineboard.pdf](https://github.com/MicrosoftDocs/windows-iotcore-docs/files/2494354/productspecification-audiomezzanineboard.pdf)

moyanming, 2018-10-19:
Hi @endyb66 ,
BTW, I recommend that you can try to use the [Audio Mezzanine Board](https://www.arrow.com/en/products/114990332/seeed-technology-limited), the price is less then $5 and free for shipping those days.

endyb66, 2018-10-19:
Thank you @moyanming
The issue with the mezzaine board is, that it does not fit in my project, because of the additional height to the Dragonboard.
Here is a demo code for recording audio `https://pastecode.xyz/view/2b278e4b`. You have to add two buttons for the eventhandlers, and further adding microphone and music library capabilities. This you you can easy download the recorded file directly from the file explorer on the Windows Device Portal inside the Music folder.
BTW: SPKR_OUT_P and SPKR_OUT_M do also work on my Dragonboard, but everything else related to audio does not. I think i read somewhere, that audio playing and recording was fine on an much older build, but i'm not exaclty shure of that.

moyanming, 2018-10-19:
Hi @endyb66 ,
I found another problem that [the volume of the Speaker can't be changed](https://github.com/MicrosoftDocs/windows-iotcore-docs/issues/749). Do you have test whether you can change the volume of your Speaker?
Best regards.

endyb66, 2018-10-20:
@moyanming, same problem here.

saraclay, 2018-10-24:
Hi @endyb66 and @moyanming, we're advising people to take a look at this guide from Qualcomm: https://developer.qualcomm.com/download/db410c/stereo-connector-and-audio-routing-application-note.pdf 

I'm closing this out for now. Feel free to reopen if the guide does not answer your questions.

closed 2018-10-24 by saraclay

endyb66, 2018-10-24:
Hi @saraclay, please reopen, this guide does not cover the problematic, which seems to be an issue with Windows 10 IoT Core.