MicrosoftDocs/windows-iotcore-docs #749: Can't set the volume of the Speaker!

moyanming, 2018-10-19:
Hi all,
DragonBoard 410c, Audio Messanine Board, test on Windows 10 IoT Core v10.0.17763.1 OS.
I have tried 3 different ways to set the volume of the Speaker, but there is no change of the Speaker. This Speaker is work fine and loudly.
1. Invalid:
 iotcoreaudiocontroltool r 58.6
2. Invalid:
Change the volume from the Audio Control of the Windows Device Portal.
3. Invalid:
Change the volume by use the button (- and +) on the Dragonboard 410c.

BTW, change the capture device volume is successfully, but it seems that the [Microphone is no work at all](https://github.com/MicrosoftDocs/windows-iotcore-docs/issues/727).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f2599338-019c-10e6-13e6-620630f42b95
* Version Independent ID: caafa3ff-dfe6-5af5-efe2-978c3469c2b2
* Content: [Release Notes for Build 17744 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/17744)
* Content Source: [windows-iotcore/release-notes/Insider/17744.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/Insider/17744.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

endyb66, 2018-10-20:
@moyanming, another way would be to change the volume in your code, on the MediaPlayer. But it seems that this is a bug what you described. A few weeks ago, i had connected speakers directly to the J7 header on the Dragonboard, and there i was also not able to change the volume with one of your three options. So it has to be a bug i think.

moyanming, 2018-10-20:
Hi @endyb66 ,
Yes, the MediaPlayer.Volum can set or get the audio volum for the media playback. This is OK for your own instance of the MediaPlayer but can't suitable for the audio which is play by the web browser.

endyb66, 2018-10-20:
@moyanming, you're right, in your use case this is not helpful. Hope we are getting news about all these analog / audio issues we have.

saraclay, 2018-10-23:
Hi - I'm trying to find the right person to help with this. Please stay tuned.

saraclay, 2018-10-24:
Hi @endyb66 and @moyanming, we're advising people to take a look at this guide from Qualcomm: https://developer.qualcomm.com/download/db410c/stereo-connector-and-audio-routing-application-note.pdf 

I'm closing this out for now. Feel free to reopen if the guide does not answer your questions.

endyb66, 2018-10-24:
Hi @saraclay, please reopen, this guide does not cover the problematic, which seems to be an issue with Windows 10 IoT Core.

saraclay, 2018-10-25:
Please file this issue on our MSDN forums here: 
https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

We explicitly say on our homepage that we handle documentation-related issues, such as typos or suggestions. However, this is falling in line with customer support, which the issues channel does not handle. 

It seems that you are working on a complex project, so it is advised that you work with a distributor. You can find a list of our distributors here: http://wincom.blob.core.windows.net/documents/Windows_IoT_Distributor_Information.pdf

closed 2018-10-25 by saraclay