MicrosoftDocs/windows-iotcore-docs #767: How to detect whether Windows 10 is headless?

christianhujer, 2018-10-24:
This is a great start to learn about Windows 10 IoT Core headless mode. It would be great to have pointers for further reading. For example, how could I, from an application, detect whether Windows 10 is in headless mode? Environment variable? System call? Where can I read more about this?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1ab72daf-549b-592f-adaf-9e0e79d6c6ca
* Version Independent ID: 523f5140-e048-0032-a2b5-5d742e2d652e
* Content: [Headed and headless devices - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/headlessmode)
* Content Source: [windows-iotcore/learn-about-hardware/HeadlessMode.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/HeadlessMode.md)
* Product: **windows-iot**
* GitHub Login: @zhuridartem
* Microsoft Alias: **artemz**

saraclay, 2018-10-25:
Hi @christianhujer, I can start you off with this -

If you want to know if headless mode is enabled or not then you can query the below registry key value for confirmation. If the headless value is set to 0x1 then device is in headless mode.
  Also as a secondary check when headless mode is enabled dwm.exe process will not be loaded. 

  <registryKey keyName="HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Wininit">
      <registryValue
          name="Headless"
          value="1"
          valueType="REG_DWORD"
          />
    </registryKey>

   <registryKey keyName="HKEY_LOCAL_MACHINE\SYSTEM\ CurrentControlSet \Control\Conkbd">
      <registryValue
          name="Headless"
          value="0"
          valueType="REG_DWORD"
          />
    </registryKey>

saraclay, 2018-10-25:
<registryKey keyName="HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Wininit">
      <registryValue
          name="Headless"
          value="1"
          valueType="REG_DWORD"
          />
    </registryKey>

   <registryKey keyName="HKEY_LOCAL_MACHINE\SYSTEM\ CurrentControlSet \Control\Conkbd">
      <registryValue
          name="Headless"
          value="0"
          valueType="REG_DWORD"
          />
    </registryKey>

saraclay, 2018-10-29:
Hi @christianhujer, I will go ahead and close this out. If you have any questions about headless (as opposed to documentation suggestions), I'd advise you ask them in the MSDN forums: https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

closed 2018-10-29 by saraclay

christianhujer, 2018-11-01:
Hello @saraclay thanks a lot for the information. I've checked whether the documentation is updated, and I couldn't find any change in the documentation. Shouldn't the information that you've posted here (registry entry, `dwm.exe` process) become part of the documentation? The page https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/HeadlessMode.md should be updated, either to include the information, or to refer another document which includes that information. Isn't it?