MicrosoftDocs/windows-iotcore-docs #774: Hi @saraclay, do your job is just to close my issues?

moyanming, 2018-10-25:
I'm closing this out for now. Feel free to reopen if the guide does not answer your questions.

_Originally posted by @saraclay in https://github.com/MicrosoftDocs/windows-iotcore-docs/issues/744#issuecomment-432720277_

moyanming, 2018-10-25:
#749 
#727

moyanming, 2018-10-25:
Where is the "I2C Accelerometer" on the github? #280
How to create a headless FFU image? #274 
How to read and write to a GPIO Pin which is not list in the GPIO Pin Table? #273

saraclay, 2018-10-25:
@moyanming - could you please file these using our Feedback Hub? GitHub issues is not a support platform. 

This is our Feedback Hub: https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT

closed 2018-10-25 by saraclay