MicrosoftDocs/windows-iotcore-docs #796: Fix your ms.prod and ms.technology metadata

msatranjr, 2018-10-26:
WHY?
We're in the process of cleaning up our Product and Technology metadata as part of a larger APEX-wide effort to get the most value from our metadata. We've established a new taxonomy so that we'll be able to filter our content in a much more meaningful way using our BI tools/reports to gain customer insights.

BACKGROUND
We use metadata on Docs for reporting, discoverability of the content via search, and to drive aspects of the site experience. For reporting, metadata particularly drives the article and master rollup reports, as well as the APEX content dashboard. Metadata can also be queried directly through Kusto. If you do not set metadata correctly, your content will be misrepresented or underrepresented in reports, resulting in less information to help you understand and serve customers.

For the site experience, metadata helps drive the search API and site search behavior and generates various site features like scoped searches, code sample highlighting, and RSS feeds. If you do not set metadata fully and correctly, Docs site features will not work properly on your content, resulting in a subpar user experience.

ACTION
If there are any instances of ms.prod and ms.technology in your markdown topics, please remove them and apply the following settings to the "globalMetadata" object in your docfx.json file.

“ms.prod” : “windows-iot”

msatranjr, 2018-10-26:
Please contact @kcpitt for more info

saraclay, 2018-10-29:
This has been done. Thanks @msatranjr!

closed 2018-10-29 by saraclay