MicrosoftDocs/windows-iotcore-docs #797: Broken Link To Get Here

RLashofRegas, 2018-10-27:
The link at the bottom of this page: https://www.microsoft.com/net/apps/iot for getting started with Windows IoT, which presumably is supposed to go here, and is broken.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f761fbd5-8a1b-e085-9e1a-8429f1ecfd79
* Version Independent ID: b912e84f-6fb4-bb81-bcc5-4c3e78441d28
* Content: [Get Started Overview - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/getstarted)
* Content Source: [windows-iotcore/GetStarted.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/GetStarted.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-10-29:
Thanks @RLashofRegas for the issue! This seems to belong to another team, so we are working with them to get this fixed. I'll close this when it has resolved.

saraclay, 2018-10-29:
Closing as this has been fixed.

closed 2018-10-29 by saraclay