MicrosoftDocs/windows-iotcore-docs #799: Cannot assign a reviewer to my pull request

tscholze, 2018-10-28:
Hi folks,
I cannot assign a reviewer to [my pull request](https://github.com/MicrosoftDocs/windows-iotcore-docs/pull/798) but I have to get the request approved. Maybe somebody could help me to get this fixed.

Thanks!

[image]

saraclay, 2018-10-29:
Hi @tscholze - apologies for not seeing this earlier. Looks like you figured it out - thank you for your contribution!

closed 2018-10-29 by saraclay

tscholze, 2018-10-29:
@saraclay thanks for merging! It's great to contribute to such a large project. 

I hope I could use this pull request as an entry to maybe win a hacktoberfest shirt :)