MicrosoftDocs/windows-iotcore-docs #809: IOT core boot failed

kamalcode, 2018-10-30:
I tried to install on raspberry 3b+,it couldn't boot .
Used 8GB SD card of MicroSD 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9df006b9-0eac-67b8-9f24-e712ec436ec8
* Version Independent ID: 491d3781-235d-1ad5-fff4-9185cbc73984
* Content: [Windows 10 IoT Core Dashboard - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/iotdashboard)
* Content Source: [windows-iotcore/connect-your-device/IoTDashboard.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/IoTDashboard.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-10-30:
Hi @kamalcode - while I don't know the specifics, you might want to try a 16 GB SD card.

saraclay, 2018-11-06:
Closing this out for now. Please open a new issue if this still is not working for you.

closed 2018-11-06 by saraclay