MicrosoftDocs/windows-iotcore-docs #826: LattePanda support

DiomedesDominguez, 2018-11-01:
LattePanda is a hackable single board computer that works beautiful with Windows IOT Core and Windows IOT Core Enterprise.

Please go to the following link for more info:
http://www.lattepanda.com/topic-f6p7413.html?sid=fc70b472a1171a3c5a0abc5479c68b4a#p7413

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3342816d-89ce-b43f-0db1-76f3d34fe1cc
* Version Independent ID: 37cec018-8b1b-379a-4392-f1c4ddbfca86
* Content: [Downloads - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/downloads)
* Content Source: [windows-iotcore/Downloads.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/Downloads.md)
* Product: **windows-hardware**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-11-15:
Thanks @DiomedesDominguez. Is there anything in particular you'd like us to do with this?

saraclay, 2018-11-20:
Hi @DiomedesDominguez, do you mind opening a PR for this? Just click on "Edit" in the upper right hand of the docs to submit it. I can take a look once it's submitted. Thanks! Closing this out.

closed 2018-11-20 by saraclay