MicrosoftDocs/windows-iotcore-docs #847: Broken links

pangar-ms, 2018-11-09:
Link to Hello World sample  is broken.
In addition it is not clear that i can run this app on desktop or has to run on IoT device only. 
It is not working on Windows Desktop i.e. on F5 Run is not hit at all and if i try to break VS says "no process is being debugged"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bfeba98b-ce89-613d-f31e-400faf6eb8c2
* Version Independent ID: 6aa7fea5-8c12-3306-4a20-f9ed8dc21344
* Content: [Background Applications - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/backgroundapplications)
* Content Source: [windows-iotcore/develop-your-app/BackgroundApplications.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/BackgroundApplications.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-11-09:
HI @pangar-ms, thank you for letting us know about the broken link - I've taken care of that. 

Regarding F5, are you deploying to a remote device?

pangar-ms, 2018-11-09:
No, I was trying on local machine.

saraclay, 2018-11-09:
Ok, you will need to deploy to a remote device.

closed 2018-11-09 by saraclay