MicrosoftDocs/windows-iotcore-docs #907: Remote Debugging does not work

ThumbWhereJames, 2018-11-21:
Remote Debugging from Vs 2017 Community (All latest patches) does not work. Can deploy but cannot debug, despite manually launching debugger via 'Device Portal' and disabling firewall. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: afdf6b9f-3d73-a1f2-4b57-684c07cacdab
* Version Independent ID: d9f4a048-6942-e360-933c-e4d11f4b56bb
* Content: [Release Notes for Raspberry Pi 3B+ - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/rpi3bp)
* Content Source: [windows-iotcore/release-notes/Insider/RPi3BP.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/Insider/RPi3BP.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

saraclay, 2018-11-26:
Hi @ThumbWhereJames - is this specifically for the RPi3B+? Or another device?

saraclay, 2018-12-03:
Closing for now. @ThumbWhereJames - please open again if this is a blocking issue for you.

closed 2018-12-03 by saraclay