MicrosoftDocs/windows-iotcore-docs #928: Can USB Wifi Adapters Be Used?

EnderNightLord-ChromeBook, 2018-12-01:
I have just a normal usb wifi adapter that can be used on all OS's (linux, macOS and windows 10) does this work with iot core? 

(using a raspberry pi 2)

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 563b22e3-1c62-eaad-eb29-6f95b01316ba
* Version Independent ID: 385c19a8-007a-f000-496c-0521606cbf9c
* Content: [Overview of Windows 10 IoT Core - Windows IoT](https://docs.microsoft.com/en-gb/windows/iot-core/windows-iot-core)
* Content Source: [windows-iotcore/windows-iot-core.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/windows-iot-core.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-12-03:
Hi @EnderNightLord-ChromeBook , 

You can verify with the list here:
https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/hardwarecompatlist

If what you have is not listed, I would still encourage you to give it a try and, if it works, please submit a PR to contribute to the document and let others know that it's also supported. As mentioned in the article, the list is not exhaustive.

closed 2018-12-03 by saraclay

EnderNightLord-ChromeBook, 2018-12-05:
sounds good thanks!