MicrosoftDocs/windows-iotcore-docs #943: Raspberry Pi 3B+ rather than 3B.

skipfire, 2018-12-12:
Per the release notes I am posting here for 3B+ use cases where 3B is not viable.  

I have some locations where a PoE capable ethernet jack is present, but no power outlet within 20+ feet which makes powering through means other than PoE not feasible.  I have these scenarios in industrial, commercial, and residential spaces where ceiling mounted equipment is the preferred option.  Trying to accomplish the same with a 3B requires much more physical space for external PoE adapters, or PoE HAT adapters that require additional cable to bridge the Ethernet to the port after the power has been ejected into the HAT.

I could get by without the WiFi for these use cases as I get data over the PoE connection as well, but I do need support for the official display and bluetooth.

saraclay, 2018-12-12:
@skipfire let me get back to you on this.

chsha, 2018-12-12:
@skipfire Thank you for the feedback. Unfortunately, there are subsystem support limitations that prevent use of the onboard wifi/BT support on the 3B+. 3rd party wifi/BT dongles do work though I understand that's a sub-par option for cost and other reasons. The RPi display should work with 3B+, but it will use software rendered graphics similar to the 3B.

closed 2018-12-14 by saraclay