MicrosoftDocs/windows-iotcore-docs #944: Serial UART

umdomby, 2018-12-12:
https://stackoverflow.com/questions/53735301/uart-raspberry-pi-iot-windows-iot-core
On the Internet there is no normal working code for receiving data on the UART, could you write a code more working?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 525c643a-0e4f-a732-cc19-bc304acc5862
* Version Independent ID: 028d320a-4eb6-a419-4933-b64c7abdf145
* Content: [Raspberry Pi 2 & 3 Pin Mappings - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/pinmappings/pinmappingsrpi)
* Content Source: [windows-iotcore/learn-about-hardware/PinMappings/PinMappingsRPI.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/PinMappings/PinMappingsRPI.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

umdomby, 2018-12-12:
Working code for communication UART (RX TX) arduino and Raspberry pi
It works through a voltage converter 5V -> 3.3V. RX TX do not forget to swap.

        public sealed partial class MainPage : Page
    {
        private SerialDevice SerialPort;
        private DataWriter dataWriter;
        private DataReader dataReader;
        string rxBuffer;     
        CancellationTokenSource ReadCancellationTokenSource = new CancellationTokenSource();

        public MainPage()
        {
            this.InitializeComponent();
            InitSerial();   
        }

        private async void InitSerial()
        {
            string aqs = SerialDevice.GetDeviceSelector("UART0");
            var dis = await DeviceInformation.FindAllAsync(aqs);
            SerialPort = await SerialDevice.FromIdAsync(dis[0].Id);
            SerialPort.WriteTimeout = TimeSpan.FromMilliseconds(1000);
            SerialPort.ReadTimeout = TimeSpan.FromMilliseconds(1000);
            SerialPort.BaudRate = 9600;
            SerialPort.Parity = SerialParity.None;
            SerialPort.StopBits = SerialStopBitCount.One;
            SerialPort.DataBits = 8;

            dataWriter = new DataWriter();
            //dataReader = new DataReader(SerialPort.InputStream);
            Listen();
        }

        public async void SerialReceived()
        {
            /* Read data in from the serial port*/
            const uint maxReadLength = 1024;
            dataReader = new DataReader(SerialPort.InputStream);
            uint bytesToRead = await dataReader.LoadAsync(maxReadLength);
            rxBuffer = dataReader.ReadString(bytesToRead);
            receivedData.Text = rxBuffer;
        }

        public async void SerialSend(string txBuffer2)
        {
            /* Write a string out over serial */
            string txBuffer = txBuffer2;
            dataWriter.WriteString(txBuffer);
            uint bytesWritten = await SerialPort.OutputStream.WriteAsync(dataWriter.DetachBuffer());
        }

        private void Button_Click(object sender, RoutedEventArgs e)
        {
            SerialSend("Hello Serial");
        }

        private async void Listen()
        {
            try
            {
                if (SerialPort != null)
                {
                    dataReader = new DataReader(SerialPort.InputStream);

                    // keep reading the serial input
                    while (true)
                    {
                        await ReadAsync(ReadCancellationTokenSource.Token);
                    }
                }
            }
            catch (Exception ex)
            {
                throw new Exception("Uart Error", ex);
            }
        }

        private async Task ReadAsync(CancellationToken cancellationToken)
        {
            Task<UInt32> loadAsyncTask;
            uint ReadBufferLength = 1024;
            // If task cancellation was requested, comply
            cancellationToken.ThrowIfCancellationRequested();
            // Set InputStreamOptions to complete the asynchronous read operation when one or more bytes is available
            dataReader.InputStreamOptions = InputStreamOptions.Partial;
            // Create a task object to wait for data on the serialPort.InputStream
            loadAsyncTask = dataReader.LoadAsync(ReadBufferLength).AsTask(cancellationToken);
            // Launch the task and wait
            UInt32 bytesRead = await loadAsyncTask;
            if (bytesRead > 0)
            {
                receivedData.Text = dataReader.ReadString(bytesRead);
            }
        }
    }

saraclay, 2018-12-12:
Hi @umdomby - could you post this question on our forums, here? https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT 

We have a team that monitor the forums and would be better equipped to help you out with an ask like this. We usually only handle questions related to documentation issues as opposed to technical issues.

Thank you!

closed 2018-12-12 by saraclay