MicrosoftDocs/windows-iotcore-docs #956: Windows Device Portal not working after latest Windows Update

JeffMushmo, 2018-12-14:
Our devices running Windows IoT Core seem to be doing their automatic updates. After the update is complete, the Windows Device Portal mentioned here: https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/deviceportal is no longer accessible. The error I get in my browser is INET_E_RESOURCE_NOT_FOUND.

These were accessible before and are our primary way of maintaining several devices.

Failing OS Version number:
10.0.17763.194

Working OS Version number:
10.0.17134.407
10.0.17134.471

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3109c64e-0fff-04f4-d572-84851f558643
* Version Independent ID: 2db48f5c-f8c6-4258-eb6f-1e0710c014f0
* Content: [Windows Device Portal - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/deviceportal)
* Content Source: [windows-iotcore/manage-your-device/DevicePortal.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/manage-your-device/DevicePortal.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

HXode, 2018-12-17:
I have the same situation as you.

saraclay, 2018-12-17:
Yikes! Thanks for pointing this out. A few things...

- Can you file this via the Feedback Hub? Instructions to do so are [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT).
- In the meantime, I will pass this along to the team immediately. Thank you.

JeffMushmo, 2018-12-17:
Already reported in Feedback Hub a couple days before posting here. Never received any counter-feedback or acknowledgement. As I'm sure you can understand, this is extremely urgent as it cripples our production.

Just got news that it also is keeping the Device Management client from establishing a connection with Azure IoT Hub. Which means we are no longer able to push apps to any of our devices (running 10.0.17763.194) remotely.

We've got devices piling up that aren't able to make it through their final production stamp-of-approval.

**Is there any known way we can rollback Windows Update to a previous version?**

MickeyH71, 2018-12-17:
We have about 20 out of 200 devices that are down after latest Windows Update 10.0.17763.194.  On the devices that are down, the Device Portal cannot be accessed over port 8080 and our background applications are not working.

udde, 2018-12-18:
I too am very interested in getting my devices working again

saraclay, 2018-12-18:
@MickeyH71 - were the other 180 devices upgraded? Or not?

MickeyH71, 2018-12-19:
We spot checked them but did not check them all. most of the ones we checked had upgraded successfully and a few had not started the upgrade yet. We think about 15-20% of the devices which applied the update failed after the update

saraclay, 2018-12-19:
Interesting. This is helpful information. Thank you @MickeyH71, I will pass this on to the team.

honorew, 2018-12-24:
I also have this situation .

pradeep3093, 2018-12-28:
I think problem is due to timing.. it was disconnected in 2-5min i think.. after that need to disconnect and connect wifi again.. then it was connected.. it is the bug in the os or wifi support driver

jorisvergeer, 2018-12-28:
I have the same problem. 

It appears that the latest update bricks the RPI. Have an 8GB SD card. Maybe its running out of available space during update.

Nonetheless. It's quite annoying. I cannot do anything with my device online, without it blowing itself up.

saraclay, 2019-01-02:
@jorisvergeer It's possible. It's recommended ot use a 16 GB card just in case. With complete transparency, the holidays slowed down investigation so I'm still pushing this bug.

saraclay, 2019-01-07:
We are not able to repro this. If someone can provide more thorough and detailed repro steps via the Feedback Hub, that will be much more helpful to us. Instructions on how to do so can be found here: [](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT )

closed 2019-01-07 by saraclay

JeffMushmo, 2019-01-07:
I have over 100 devices I can consistently repro this on.

JeffMushmo, 2019-01-07:
I just tried submitting the logs through the automatic process. But now, every time I launch IoT Dashboard, the app hangs and says (Not Responding).

The lack of stability of the entire platform is becoming a serious concern for the business and is causing us to re-evaluate our options.

albi90, 2019-01-08:
I also have about 30 devices doing this, disabling the firewall allows device portal to work again but im getting an alot of 500 errors when trying to use the API for basic things like screenshots

albi90, 2019-01-08:
As noted above when using a freshly built image all is good but updated devices are the ones with issues, im using Intel (fitlet2) hardware not RPI.

HXode, 2019-01-09:
PowerShell :
    netsh firewall set opmode disable
    net stop webmanagement
    net start webmanagement
Device portal works again. Did the firewall update to 17763 make webmanagement not work??

JeffMushmo, 2019-01-09:
Attempting to attach the logs from the IoT Dashboard, doesn't work.

In addition, nobody has responded to the feedback within feedback hub.

Based on this and the number of people still having problems. Please re-open the issue.

JeffMushmo, 2019-01-09:
This issue also keeps us from remotely deploying apps to IoT Core from Visual Studio.