MicrosoftDocs/windows-iotcore-docs #957: These updates are causing havoc!

alsliahona, 2018-12-16:
Several years ago I built a pool pump controller using a Raspberry Pi and Windows 10 IoT.  It has a 5" LCD display and allows me to set timers for my pool pumps.  Which is awesome.  Except when you push an update out to the Raspberry Pi in the middle of winter, and I need to take the whole thing apart to plug-in a keyboard, so that I can login to give authorization for data collection!  In the meantime I have to pray that there are no freezes until I can get this thing working again.  I'm staring to regret the Windows 10 IoT choice now.

How about an option that says "This device is working as expected, don't update it!" ?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e550244c-f4be-049b-9131-ba32bd3e9214
* Version Independent ID: 77ea5cc8-8a98-53e4-affd-a1f30538bee6
* Content: [October 2018 Update - Build 17763 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/currentcommercial)
* Content Source: [windows-iotcore/release-notes/CurrentCommercial.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/CurrentCommercial.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

endyb66, 2018-12-16:
Isn't it possible to autologin inside your app, or using a touch screen? Or didn't you mean autorization inside your app?

alsliahona, 2018-12-16:
> Isn't it possible to autologin inside your app, or using a touch screen? Or didn't you mean autorization inside your app?

No, not within the App.  When they push an update, the app no longer auto-runs (may not even be on the device -- I haven't checked yet), and the device auto-logs on, but they are no presenting a screen for allowing uploads to Microsoft, which you can't opt out of and which requires a login.

endyb66, 2018-12-16:
I don't know, if i understand you correctly, but when you set an app as default (headed or headless), this app does autostart again, as soon as the windows update is done (after device rebboot and installation). So this should not be a problem.

alsliahona, 2018-12-17:
That is definitely not what happened.  In fact, there is no evidence that my app is even installed on the device.  I'll be taking it apart and bringing it in the house later today to re-install the app.  If I had time (I really don't have time for this even), I would consider switching to something else.  This is the 3rd time I've had to do this since building the device several years ago.

endyb66, 2018-12-17:
hm, that's weird because one of my devices is running since nearly two years with the same headed default app and it "survived" every OS update.

saraclay, 2018-12-17:
Hi @alsliahona - we received similar feedback from another user and we are looking into this. 

To help us repro this issue, can you let us know if you uninstalled the "Default App" from the Windows Device Portal (WDP)? It would be helpful to us if you can take a screenshot of the "Apps" section from WDP. Thank you.

alsliahona, 2018-12-17:
@saraclay No, I believe I installed it from Visual Studio 2017.
When I restarted the device just now, with a keyboard attached, it asked for Windows Device Portal credentials -- which seems to be something new -- or it has been long enough since I last worked on the device that I have forgotten about it.

alsliahona, 2018-12-17:
@saraclay 
[image]

[image]

saraclay, 2018-12-19:
Hi @alsliahona - my apologies for the poor wording. I wanted to clarify that we do have a bug on our team for this and we are seriously looking into this and trying to get to the bottom of this. I definitely understand your concern here and we are not taking this issue lightly.

saraclay, 2019-01-07:
If this happens again, could you please take note of how to reproduce this? Devices are set up in different ways, running different images, etc. and it can be really difficult for us to reproduce the error. Instructions on how to send this sort of feedback to the Feedback Hub can be found here: https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT 

Thank you so much.

closed 2019-01-07 by saraclay