MicrosoftDocs/windows-iotcore-docs #958: Flashing with eMMC for DragonBoard 410c

emanueleleoni, 2018-12-17:
To complete the installation of Iot Core, after step 5 and before step 6, you have to add another step where you have to click again "Browse" and select "rawprogram0.xml" generated in step 5 and then click the "Program" button.
Only with this step the download process begins.
I hope this can helps.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6547b7e1-4e53-9510-e550-ac2d369da6d9
* Version Independent ID: a3c18768-7ee4-4e7a-2ca4-f6f6c31f342a
* Content: [Setting up your device - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/quickstarter/devicesetup)
* Content Source: [windows-iotcore/tutorials/quickstarter/DeviceSetup.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/tutorials/quickstarter/DeviceSetup.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-12-17:
Hi @emanueleleoni, thank you so much for this feedback.

Could you take a look at the PR I created and see if this aligns to your thinking?
https://github.com/MicrosoftDocs/windows-iotcore-docs/pull/960

emanueleleoni, 2018-12-18:
Hi @saraclay ,
I think it's all right. Only one thing, when you write
"The nclick the "Program" button" 
The letter "n" is near "click" instead of "the"

Thank you very much.
Emanuele

saraclay, 2018-12-18:
Good catch. Thank you!

closed 2018-12-18 by saraclay