MicrosoftDocs/windows-iotcore-docs #962: Raspberry Pi 3B+

Trenticus, 2018-12-19:
Great work on Windows IoT. I am planning to buy a Raspberry Pi 3B+ soon. I know Windows IoT support for 3B+ is in technical preview and does not yet support Onboard WIFI and Bluetooth. Do you have any time estimate as to when there will be support for 3B+ or would there perhaps be a forthcoming technical preview (version 2) that might support onboard wifi and bluetooth on the 3B+?  Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 257dc32b-f302-db2a-2571-8f217007d893
* Version Independent ID: 92b2f238-f621-3feb-5b5b-07aed095b115
* Content: [Suggested Prototype Boards - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/quickstarter/prototypeboards)
* Content Source: [windows-iotcore/tutorials/quickstarter/PrototypeBoards.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/tutorials/quickstarter/PrototypeBoards.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-12-19:
Hi @Trenticus, we would highly recommend using the 3B for commercialization. There is no timeline for a version 2. In the future, we are looking to work closely with the Raspberry Pi foundation so that we can better support the next series of RPi. Is there a reason that using the 3B would be a blocker issue?

Trenticus, 2018-12-19:
Thanks for the reply. No, using 3B is not a blocker issue... that's the direction we will go. If support for the 3B+ was imminent we would simply use their latest hardware but we definitely can use the 3B for now.

closed 2018-12-19 by Trenticus