MicrosoftDocs/windows-iotcore-docs #968: Unable to Deploy or Remote Connect to my Raspberry Pi 3

VitalLogic, 2018-12-25:
I have installed on Sandisk 16GB Class 10 card, IoT v.10.0.17763.107, on RPi 3, and It boots good. I can also see the device on Dashboard, and Device portal, working from Windows 10 Pro. However the following issues are observed. 
1. Visual Studio, Universal application, identifies my remote system, but rejects to deploy in following configurations 
a) ARM with Universal (Unencrypted Protocol), 
Severity	Code	Description	Project	File	Line	Suppression State
Error		DEP6957: Failed to connect to device '192.168.1.12' using Universal Authentication. Please verify the correct remote authentication mode is specified in the project debug settings. COMException - Exception from HRESULT: 0x801C0003 [0x801C0003]	App1			
b) ARM - Windows/None mode with following error 
Error: Unable to connect to the Microsoft Visual Studio Remote Debugger named 'KisnaProto'.  The Visual Studio 2017 Remote Debugger (MSVSMON.EXE) does not appear to be running on the remote computer. This may be because a firewall is preventing communication to the remote computer. Please see Help for assistance on configuring remote debugging.

2) My Windows IOT Remote client doesn't connect to the host, nor throws any error , just white blank screen, after few seconds of "proccessing-rotating icon" 
Help is appreciated. I have enabled IPV6, 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 563b22e3-1c62-eaad-eb29-6f95b01316ba
* Version Independent ID: 385c19a8-007a-f000-496c-0521606cbf9c
* Content: [Overview of Windows 10 IoT Core - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/windows-iot-core)
* Content Source: [windows-iotcore/windows-iot-core.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/windows-iot-core.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-01-02:
Hi @VitalLogic - is this a Windows 10 IoT Core question? You said Windows 10 Pro so I'm a little confused. Could you please clarify?

saraclay, 2019-01-07:
I am assuming this question was for Windows 10 Pro, so I will close this out.

closed 2019-01-07 by saraclay

yobl, 2019-09-06:
Hello,
I have also this problem on RPI3B.
Error DEP6957.
I use VS 2019 version 16.2.4 and Windows 10 PRO 1903 version 18362.329.
My Raspberry PI 3B has version 10.0.17763.678.
Why is it so difficult to develop with Win iot ???
I reinstall UWP module but without any amelioration.
Hope someone can help.