MicrosoftDocs/windows-iotcore-docs #970: IoT Core Dashboard installer appears unsigned

philmea, 2018-12-28:
Hi,
The installer for the IoT Core Dashboard, setup.exe, appears to not be signed. This blocks installation on many corporate machines and prompts a warning that the installer is potentially harmful. Tested on Windows 10. Product feedback hub was unavailable.

[image]

@riverar - GitHub issue to track your feedback. Thanks for the report!

Phil
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9df006b9-0eac-67b8-9f24-e712ec436ec8
* Version Independent ID: 491d3781-235d-1ad5-fff4-9185cbc73984
* Content: [Windows 10 IoT Core Dashboard - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/iotdashboard)
* Content Source: [windows-iotcore/connect-your-device/IoTDashboard.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/IoTDashboard.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

riverar, 2019-01-02:
@saraclay Can you add more information on the closure? Is it fixed?

saraclay, 2019-01-02:
Apologies @philmea @riverar - we are investigating this. Stay tuned.

saraclay, 2019-01-11:
Hi @philmea @riverar - this issue has now been resolved.

closed 2019-01-11 by saraclay