MicrosoftDocs/windows-iotcore-docs #972: Visual Studio do not contain window like here

VictorDenisenko, 2019-01-01:
Visual Studio 2017 and 2015 do not contain window like it is showed here below sentence
"In Visual Studio, you can configure your target by editing your project's properties (be sure to make all of the highlighted changes as appropriate to your board's name or IP address):"
Can you give string path to this windows in VS ?

---
#### Сведения о документе

⚠ *Не вносите правки в этот раздел. Это необходимо для связывания страницы сайта docs.microsoft.com с вопросом на GitHub.*

* ID: 6c49bd94-e4f5-da54-37b5-05e2c8556f2d
* Version Independent ID: 6987221f-5acd-c9d8-cf7d-96d6264a272a
* Content: [Debug your app using Remote Console App Debugging - Windows IoT](https://docs.microsoft.com/ru-ru/windows/iot-core/develop-your-app/remotedebugging)
* Content Source: [windows-iotcore/develop-your-app/RemoteDebugging.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/RemoteDebugging.md)
* Product: **windows-iot**
* GitHub Login: @bfjelds
* Microsoft Alias: **bfjelds**

bfjelds, 2019-01-02:
We can work to clarify this … to unblock you, that window location is described [here](https://docs.microsoft.com/ru-ru/windows/iot-core/develop-your-app/appdeployment) … it is in the Project Properties, which can be accessed from the Solution Explorer context menu.

bfjelds, 2019-01-02:
Here is more info on the Project Properties dialog: https://docs.microsoft.com/ru-ru/visualstudio/ide/managing-project-and-solution-properties?view=vs-2017

saraclay, 2019-01-02:
I opened a PR for this. @bfjelds does this look good to you? https://github.com/MicrosoftDocs/windows-iotcore-docs/pull/974

VictorDenisenko, 2019-01-03:
I think, I cannot find it because I use UWP C# background app, it's Project property do not contain windows like here. So, I'm sorry to bother you.

closed 2019-01-03 by VictorDenisenko