MicrosoftDocs/windows-iotcore-docs #981: receipt printer

huseyinyildizHYC, 2019-01-04:
We use windows 10 iot core and raspbery pi 3 on an industrial project. And we build a machine with it. 
We should use a receipt (thermal or pos) printer. But we cant find simple solution.  Are there any available working driver. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9c8da9d0-10ee-b8e6-6c11-16f1faf8a630
* Version Independent ID: e45ebd71-6334-7dd6-87a2-933210996f4c
* Content: [Hardware Compatbility List - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/hardwarecompatlist)
* Content Source: [windows-iotcore/learn-about-hardware/HardwareCompatList.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/HardwareCompatList.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-01-07:
Hi @huseyinyildizHYC, could you elaborate? Please note that the Hardware Compatibility list is a limited list - there are many, many peripherals out there and we are not able to test everything that's out there.

huseyinyildizHYC, 2019-01-08:
Hi Sara,

Thank you for your response.

We are using Raspberry Pi 3 as controller and a windows IOT Core based software in our industrial projects. In one of our recent projects we need our machine to give printed reports. So we decided to use a thermal printer. Our problem is that, we could not find a thermal printer we can use on our machine. Because we could not find drivers for thermal printers on windows IOT compatible with  ARM processors.  

Could you recommend us a thermal printer that has a driver compatible with windows IOT Core and ARM processor or could you provide us a driver for the thermal printer that we choose.

Best regards.

Huseyin Yildiz
Electronics Engineer

saraclay, 2019-01-11:
Hi @huseyinyildizHYC, here's some info from the team:

With Windows 10 IoT Core, I believe the only thing we can support today is either a Bluetooth or network connected thermal receipt printer.  An easy way to do this might be to connect a network thermal receipt printer directly to the RPi.  In this case, the thermal receipt printer must be ESC/POS compatible (see here: https://aka.ms/pointofservice-hw).

If you're having some issues using the most the latest build of Windows 10 IoT Core, you may want to try the April 2018 update, available here (https://docs.microsoft.com/en-us/windows/iot-core/downloads).

closed 2019-01-14 by saraclay