MicrosoftDocs/windows-iotcore-docs #982: Didn't work, no wireless connection profile found

FinitelyFailed, 2019-01-07:
This examples doesn't work, I cannot find any wireless connection profiles. I found this examples that does work: http://studio-geek.com/archives/1031.

var tetheringManager = NetworkOperatorTetheringManager.CreateFromConnectionProfile(NetworkInformation.GetInternetConnectionProfile());

string AP_SSID = "network";
string AP_PASS = "fubarfubar";
var accessPointConfig = new NetworkOperatorTetheringAccessPointConfiguration()
{
    Ssid = AP_SSID,
    Passphrase = AP_PASS
};

await tetheringManager.ConfigureAccessPointAsync(accessPointConfig);
await tetheringManager.StartTetheringAsync();

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4b9db9be-5d4c-611a-674e-c463859a03b5
* Version Independent ID: a3729731-8407-e6b4-891f-46235f26c299
* Content: [Internet Connection Sharing - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/internetconnectionsharing)
* Content Source: [windows-iotcore/connect-your-device/InternetConnectionSharing.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/InternetConnectionSharing.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-01-07:
Hi @FinitelyFailed - I am investigating this. Stay tuned.

saraclay, 2019-01-17:
Sorry for the delay, @FinitelyFailed. We were able to get around to testing this and have confirmed that it doesn't work for us either. Now to figure out why...

FinitelyFailed, 2019-01-18:
With the code above I get tethering to work, and I can connect to the iot device from a laptop. However after a time, around 1 min, the laptop gets disconnected. When I reconnect the same thing happens again and again. At some point the tethering turned it self off by it self, quite randomly. I don't know where to report this. And I have a hard time finding documentation and examples for this, as well as most things concerning Windows 10 IoT Core app development ...

saraclay, 2019-01-22:
Hi @FinitelyFailed, I'm sorry to hear about your experiences with our documentation. I've sent your additional feedback to someone on our team. While they work on it, I'd like to learn more about what else you're having trouble finding in general. Looking to improve where we can based on user feedback.

saraclay, 2019-01-29:
Hi @FinitelyFailed, first of all, thank you so much for the feedback. While we're restrained with our docs platform, that's no excuse for delivering a poor experience. I'll work to see what we can do for that.

Regarding your question at hand, here are a few docs that might help if you haven't seen them yet:
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/backgroundapplications 
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/onscreenkeyboard

saraclay, 2019-02-08:
Hi @FinitelyFailed, we were able to solve this and it looks like 1) you may have created a WiFi Profile at first and 2) you will need to add this to your manifest: 
<DeviceCapability Name="wiFiControl" />    

Hope this works for you.

closed 2019-02-13 by saraclay