MicrosoftDocs/windows-iotcore-docs #989: stuck on rainbow screen Raspberry pi 3B+

quintendc, 2019-01-09:
Hello,

i tried to run Windows IoT Core in my raspberry Pi 3B+ but i got stuck on the rainbow screen.
i used a 32GB Micro SD card:
https://www.amazon.de/gp/product/B073JWXGNT/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

i also tried the 128 GB version but no luck 
https://www.amazon.de/gp/product/B073S8TQH6/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

is the raspberry Pi 3B+ not supported? or does it only support 16 GB Version?

thanks in advance,

saraclay, 2019-01-09:
Hi @quintendc, there are a few things you can try...

1. Can you confirm that you downloaded the image from this site: https://www.microsoft.com/en-us/software-download/windowsiot 

2. Can you confirm your steps with the steps outlined here: https://www.hackster.io/JiongShi/windows-10-iot-core-for-raspberry-pi-3-model-b-92b1a3 

3. If neither of the two steps work for you, please take a look at the thread in the forum: https://social.msdn.microsoft.com/Forums/en-US/5239e325-f2c6-435c-9f83-ad07b6f7d757/raspberry-pi-3-b-stuck-on-rainbow-screen-after-installing-windows-iot?forum=WindowsIoT

quintendc, 2019-01-10:
Ok thank you that worked.

closed 2019-01-10 by quintendc