MicrosoftDocs/windows-iotcore-docs #1049: GPIO 25 no longer available

alexiordan, 2019-01-27:
According to this: https://social.msdn.microsoft.com/Forums/lync/en-US/29cd77f8-4c8f-4baf-9ebf-9d14aa066009/the-gpio25-of-the-dragonboard-410c-cant-be-opened?forum=WindowsIoT , it is acknowledged by Microsoft that GOIO 25 is no longer available since lack of support from Qualcomm BSP.
It will be nice to also state from what version of Windows 10 IoT is the GPIO 25 unavailable.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cf4b39be-e0ef-9db7-d542-248e8d107471
* Version Independent ID: 2862e930-3f9f-c73f-cf57-1396ceeeadc0
* Content: [Dragonboard Pin Mappings - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/pinmappings/pinmappingsdb#gpio-pins)
* Content Source: [windows-iotcore/learn-about-hardware/PinMappings/PinMappingsDB.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/PinMappings/PinMappingsDB.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-01-29:
GPIO 25 was never supported.

closed 2019-01-29 by saraclay

alexiordan, 2019-01-30:
Hi @saraclay, have you read the thread at the posted link?

moyanming, 2019-01-30:
Hi @alexiordan ,
The GPIO 25 should be supported by the IoT OS, but it seems that this issue still not resolved by the 1809.