MicrosoftDocs/windows-iotcore-docs #1056: 3B+ technical preview

pk386, 2019-01-31:
When is the stable release for the 3+ going to be available?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9df006b9-0eac-67b8-9f24-e712ec436ec8
* Version Independent ID: 491d3781-235d-1ad5-fff4-9185cbc73984
* Content: [Windows 10 IoT Core Dashboard - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/IoTDashboard)
* Content Source: [windows-iotcore/connect-your-device/IoTDashboard.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/IoTDashboard.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-02-01:
@pk386 We do not have a release date, but likely around late fall 2019.

pk386, 2019-02-02:
@saraclay Thanks, 
I'm working on a project requiring the wi-fi capabilities. but we can rely on the built in Ethernet until later this year.

Is there a way I can be notified of the stable release?

Thanks in advance.

saraclay, 2019-02-04:
@pk386 If you're a Windows Insider for Windows IoT, you will likely hear that way. Also, you'll probably hear about it if you're subscribed to the Microsoft IoT newsletter, which you can sign up for here: https://profile.microsoft.com/RegSysProfileCenter/wizardnp.aspx?wizid=78930ef8-d176-4b5e-b49f-2cefcb6b0891&lcid=1033

closed 2019-02-04 by saraclay