MicrosoftDocs/windows-iotcore-docs #1079: unidad U: da error

stanzianivm, 2019-02-08:
Hola, quisiera ver si me pueden ayudar con un problema que estoy teniendo. 

Sigo todos los pasos instalando los package de la version de windows iot core 17763, hago el commit y despues introduzco el comando uwfmgr.exe filter enable para poder proteger las unidades C: y U:.

Con al unidad C: lo protege sin problemas, pero cuando quiero proteger la unidad U: me tira una pantalla azul que dice 
"your pc run into a problem that it couldn't handle and now it needs to restart"

Desde ya muchas gracias.

Saludos, Victor.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bd4b07ee-821f-42bf-5fc8-c330e8dae76f
* Version Independent ID: 66f8fd67-1c58-914a-c267-f0c22812fc1d
* Content: [Using the Unified Write Filter - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/secure-your-device/unifiedwritefilter)
* Content Source: [windows-iotcore/secure-your-device/UnifiedWriteFilter.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/secure-your-device/UnifiedWriteFilter.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-02-12:
Gracias por su pregunta @stanzianivm. Pediré al equipo pronto...

saraclay, 2019-02-13:
Hola @stanzianivm - una respuesta, pero in inglés:

We need a bit more information to understand the issue... 
_Necesitamos más información_

Is it possible to get kernel dump for that bugcheck? - customer will need to connect kernel debugger and collect the dump...
_Es posible para obtener el kernel dump para el bugcheck? La persona necesitará conectar el kernel debugger y sacar el dump_

In meantime can we get output of uwfmgr get-config command after U is protected but right before the reboot.

saraclay, 2019-02-21:
Cerraré esto.

closed 2019-02-21 by saraclay