MicrosoftDocs/windows-iotcore-docs #1080: Installation did not work

lancedecker, 2019-02-11:
I've tried this multiple times and I am not getting Windows 10 IoT to boot.  I get the four color screen, but it never moves beyond this.  I have successfully installed Raspian on a different SD and it runs properly.  I have gone all the way back and deleted all the partitions and formatted the SD card.  The card is a 32 GB card.  The last time, I did not get any failures on the IoT Dashboard durin the installation process.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3342816d-89ce-b43f-0db1-76f3d34fe1cc
* Version Independent ID: 37cec018-8b1b-379a-4392-f1c4ddbfca86
* Content: [Downloads - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/downloads)
* Content Source: [windows-iotcore/Downloads.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/Downloads.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-02-12:
@lancedecker, thank you for filing this issue. What image were you trying to flash? Doesn't sound like a SD card issue to me.

ashkabarnya, 2019-02-12:
Had the same issue with the latest image (17763) on PI 3B+. Taking an older image fixed the booting issue.

lancedecker, 2019-02-12:
I was using a PI 3 also.  I will try the older version.  Which version did
you use?

ashkabarnya, 2019-02-12:
I used 17661

lancedecker, 2019-02-13:
I'm having trouble finding a link to download an older version...

*Lance Decker*

saraclay, 2019-02-13:
@lancedecker what kind of SD card are you using? SanDisk has been shown to have better performance...

saraclay, 2019-02-13:
@ashkabarnya what kind of SD card are you using?

lancedecker, 2019-02-13:
I have tried several.  Samsung 64GB EVO, Sandisk Ultra 32 GB, Sandisk Ultra
128 GB.

I can get Raspian running on all of them, but when I go to Windows 10 IoT,
I always get the same failure.  I'm trying to find the older versions of
the IoT build, but cannot.  Also, I registered for the Windows Insider and
Azure.  Both seem to take 3 days to trigger.  I've heard that the Windows
Insider Preview may work...  I'm kind of in a waiting game to get the
registration completed or find an older version of the IoT build.

*Lance Decker*

--------------------------------------------------------------------------------------------------------

TheAbonimater, 2019-02-13:
Hey,

I have the exact same problem

lancedecker, 2019-02-14:
I have downloaded two preview versions but the Windows IoT dashboard is not
recognizing these two builds.  What recommendations for alternative methods
to transfer the iso to an SD card?

*Lance Decker*

saraclay, 2019-02-14:
@ashkabarnya @TheAbonimater @lancedecker 

A few more clarifying questions:

- Are you flashing on a RPi3B or B+? If the latter, this is an unsupported technical build with a release coming in a few months. For now, please stick to a RPi3B.

- If you're using a RPi3B, make sure your SD cards are wiped clean. If you've already done that, it would be great if you could file a bug in our Feedback Hub, using the instructions [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT) so we can better track it.

lancedecker, 2019-02-15:
B+ - but I need Windows 10 IoT for a project...  kind of stinks...

*Lance Decker*Corporate Program Manager

lance@bluecats.com | bluecats.com
m | +1 903-681-1447
301 Chicon St. Suite A
Austin, TX 78702

--------------------------------------------------------------------------------------------------------

saraclay, 2019-02-19:
@lancedecker are you looking to commercialize a product? Or are you just prototyping? You can still prototype with the Pi3B+, but we do not recommend it for commercializing. The 3B is recommended for commercializing.

lancedecker, 2019-02-20:
Not commercializing a product.  We are doing some research.

saraclay, 2019-02-20:
@lancedecker You can still use Pi3B+ for prototyping then. But again, if you're eventually looking to commercialize, it's probably best to use Pi3B, more because it's being used already in production.

closed 2019-02-26 by saraclay

scoaband, 2019-03-28:
You have to first SecureBoot the device before installing the flash image.