MicrosoftDocs/windows-iotcore-docs #1082: RPI 3B+ support

ashkabarnya, 2019-02-12:
Are there any plans to add full support of Raspberry PI B3+ to Windows 10 IOT Core?
When I use the latest image (17763), the system doesn't boot (rainbow screen).
When I am using an older image (17661) touchscreen (7" raspberry official touchscreen), WiFi and Bluetooth don't work.

saraclay, 2019-02-12:
Hi @ashkabarnya here's what I have for you...

-Regarding full support, I cannot say. We are planning to put out a new release in a month or two, but there still will not be onboard wifi/bluetooth.

Regarding the latest image, I believe there is a known issue for this and the team is working on it.

saraclay, 2019-02-13:
@ashkabarnya what kind of SD card are you using? Could you please provide me with size (e.g. 16 GB) and brand (e.g. SanDisk).

ashkabarnya, 2019-02-13:
Hi @saraclay, I’m using Samsung 32GB MicroSDHC EVO Select  (
https://www.samsung.com/us/computing/memory-storage/memory-cards/microsdhc-evo-select-memory-card-w--adapter-32gb--2017-model--mb-me32ga-am/ )

saraclay, 2019-02-19:
@ashkabarnya is the SD card writable? There's a possibility that it may be corrupt. Please be sure that it's been wiped clean before flashing.

oldieman, 2019-02-22:
Hi, I'm waiting for the image over 8 month now. I got the older image running but the MAC-Address of all boards are the same. Therefore it's impossible to identify the board in an IOT environment with more of the same boards (Azure IOT Hub). I need the LAN and WiFi running in short time.

saraclay, 2019-02-25:
@oldieman I'm not sure if this is something I'm able to address here. Do you mind following the steps [here ](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT )to file this to our Feedback Hub? Thanks!

closed 2019-02-26 by saraclay