MicrosoftDocs/windows-iotcore-docs #1091: Playback from iPhone to IoT device by Bluetooth is intermittent

shijiong, 2019-02-13:
Hello, friends
Today I paired my iPhone to Windows 10 IoT Core device Build 17763.107- Raspberry Pi 3B  with Bluetooth. I played music on the iPhone, and the sound from the Windows 10 IoT Core device is intermittent. I tried the onboard 3.5mm audio interface and usb sound card. However, the performance were the same. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 60f784b7-a698-cb6e-6789-6b9719970470
* Version Independent ID: df8f2bb4-9b0d-4db2-7c18-b9900ded9f19
* Content: [October 2018 Update - Build 17763 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/commercial/october2018update)
* Content Source: [windows-iotcore/release-notes/commercial/October2018Update.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/commercial/October2018Update.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-02-13:
Hi @shijiong, could you file in this in Feedback Hub using the instructions [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT)? That way, we can better track and prioritize this. Thank you!

closed 2019-02-13 by saraclay

shijiong, 2019-02-14:
Hello, Sara. @saraclay 
I have submitted the feedback by Feedback Hub.