MicrosoftDocs/windows-iotcore-docs #1092: Feature ID

JeffMushmo, 2019-02-13:
Trying to build a release image which leverages CheckNetIsolation.exe. It doesn't appear under System32. Which IoT core feature ID is needed to make sure CheckNetIsolation.exe is included with the image?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7f952bdf-550a-c165-35f2-9be9998b9eef
* Version Independent ID: 699068d8-cc46-28f2-a4b1-19138400d443
* Content: [Communicating with Localhost - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/loopback)
* Content Source: [windows-iotcore/develop-your-app/Loopback.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/Loopback.md)
* Product: **windows-iot**
* GitHub Login: @paulmon
* Microsoft Alias: **paulmon**

saraclay, 2019-02-13:
Hi @JeffMushmo, we are investigating this. Stay tuned.

saraclay, 2019-02-13:
@JeffMushmo The IOT_TOOLKIT feature ID is pulling checknetisolation.exe.

JeffMushmo, 2019-02-13:
Ok. Thank you. I get a warning when building the image that IOT_TOOLKIT is a development feature. I'm guessing it's enabling other things that I don't necessarily want. Is there a suggested way I could include the checknetisolation.exe without the entire IOT_TOOLKIT? We really like the idea of having our devices as locked-down as possible.

saraclay, 2019-02-21:
@JeffMushmo one of our devs was able to build an image without IOT_TOOLKIT and was able to use CheckNetIsolation.exe. It appears to work if you copy the .exe and .mui files over: 
c:\windows\system32\CheckNetIsolation.exe
c:\windows\system32\en-us\CheckNetIsolation.exe.mui

Let us know if this works for you.

JeffMushmo, 2019-02-21:
Great. I'm guessing for a retail image, I will need to create a CAB package that supplies these two files, and add that as an OEM feature? Or is there a better way?

saraclay, 2019-02-21:
@JeffMushmo that is correct.

I forgot to mention one important note - CheckNetIsolation.exe is not test signed. Therefore, it won’t be automatically serviced by Windows servicing if they put it in their own package. If you end up putting it in your package, you will have to watch for servicing changes in IOT_TOOLKIT and service your own package based on those changes.

JeffMushmo, 2019-02-21:
I see. Yeah, less work for us would be better. I'll wait to hear if your dev team has any other suggestions.

Seems like this CheckNetIsolation and some other utilities, like limpet.exe, are dependent on specific packages. Would be nice if we could get at these binaries without having to build an image.

Is CheckNetIsolation.exe something that I can see the source to? Maybe the functionality we need out of it could just be implemented in an app/exe that we service from our end?

Thanks

saraclay, 2019-02-25:
@JeffMushmo, I wish I could say otherwise, but this is our recommendation for the time being. Your issue has started a discussion in our dev team that may take extra work on our end, so thank you for your feedback.

saraclay, 2019-03-19:
Thank you for the update, @JeffMushmo. For further issues, please file them on our Feedback Hub so we can be sure to triage and test these issues. You can find the instructions for doing so [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT).

closed 2019-03-19 by saraclay