MicrosoftDocs/windows-iotcore-docs #1096: WiFi Profile not showing

grahamehorner, 2019-02-15:
I have an issue when running the Windows 10 IoT Dashboard, the dashboard is not showing ANY WiFi profiles regardless of having a number of WiFi profile present and working on the laptop.  I'm running Windows 10 Pro ver. 1809 build 17763.316

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 42080662-4c60-7723-061b-06c8cbf1a172
* Version Independent ID: 8db33764-e6d2-2bec-58ef-57cae635f86b
* Content: [Troubleshooting - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/troubleshooting)
* Content Source: [windows-iotcore/Troubleshooting.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/Troubleshooting.md)
* Product: **windows-hardware**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-02-19:
@grahamehorner - wha device are you trying to put Windows 10 IoT Core on? Sometimes the Raspberry Pi is unstable when it comes to WiFi - you may want to try Ethernet.

grahamehorner, 2019-02-19:
@saraclay I am attempting to create a SD card for the Raspberry PI; the Windows 10 IoT Dashboard utility that is used to create/make the bootable SD card is not detecting the WiFi on the hosting systems (Windows 10 Enterprise, Dell XPS Laptop or HP Laptop) so nothing to do with the target/IoT devices.

saraclay, 2019-02-20:
@grahamehorner it can sometimes have to do with the device. I don't see your setup, so it's hard to properly diagnose. Again, I would recommend Ethernet if this is a potential option for you. Otherwise, if you believe this is a bug, please file this on our Feedback Hub following the steps outlined [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT).

closed 2019-02-20 by saraclay