MicrosoftDocs/windows-iotcore-docs #1110: Is there Core Packages for the 17763.253?

moyanming, 2019-02-26:
Hi,
We need custom ourselves FFU, is there new Core Packages for the 17763.253?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9fa45d17-9bd5-91d0-feff-676e0ad8fb5a
* Version Independent ID: f9ef63e1-1605-02bf-1d77-b2a07ea0a80b
* Content: [Release Notes for Build 17763.253 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/commercial/17763)
* Content Source: [windows-iotcore/release-notes/commercial/17763.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/commercial/17763.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

saraclay, 2019-02-26:
@moyanming, we're in the process of getting this out. Thank you for your patience.

saraclay, 2019-03-01:
@moyanming it can be found here: https://www.microsoft.com/en-us/software-download/windows10IoTCore

closed 2019-03-01 by saraclay