MicrosoftDocs/windows-iotcore-docs #1112: Disable On Screen Keyboard

David-W-Williams, 2019-02-26:
For one of my IoT core products the keyboard is an option. 

When I have a USB keyboard plugged in, the OSK will still pop up when the Azure AD Sign in dialog appears, and the user name field is in focus.  This is annoying my end users.

Is there a reg entry or other method I can use / add to my settings app to ensure that the OSK will remain "silent" when a keyboard is plugged in?     

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 35f256a7-b5c2-0357-08fe-886e286512fa
* Version Independent ID: d16d8e75-c157-f2c8-1061-8c751c9d9a87
* Content: [Configure your headed device with the Windows 10 IoT Core on-screen keyboard - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/onscreenkeyboard)
* Content Source: [windows-iotcore/develop-your-app/OnScreenKeyboard.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/OnScreenKeyboard.md)
* Product: **windows-iot**
* GitHub Login: @JohnTasler
* Microsoft Alias: **jtasler**

saraclay, 2019-02-27:
@David-W-Williams Can you provide me with the OS version #?

saraclay, 2019-03-07:
Closing as there has been no customer follow-up. Please open a new issue if this is still not working.

closed 2019-03-07 by saraclay