MicrosoftDocs/windows-iotcore-docs #1120: No download link

AndrewBracher, 2019-03-02:
How is a developer, who is in the 'insider' program, supposed to evaluate IOT Enterprise for developing his Kiosk app, if there is no download and no explanation of how to go about this.  I can see Core is there, but nothing on Enterprise.  Not even a release schedule.  Baffling!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a1e7dc7e-3df6-2eaf-9c11-ed03c00719e0
* Version Independent ID: ab5e43cf-c4db-cd8f-e92a-c33e4ebc2b68
* Content: [Overview of Windows 10 IoT Enterprise - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/windows-iot-enterprise#feedback)
* Content Source: [windows-iotcore/windows-iot-enterprise.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/windows-iot-enterprise.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-03-05:
Thank you for the feedback, @AndrewBracher. I will work with the team to see how we could improve this experience.

closed 2019-03-05 by saraclay

AndrewBracher, 2019-03-05:
Hmmm like that's going to happen...

esammann, 2019-04-17:
I would also like to know where to find a download of Windows 10 IoT Enterprise.  It really seems like this should not be a closed topic since nothing has been added to give us direction on this.

esammann, 2019-04-17:
So I see in one of the comments below that there is a link, but I guess I'm not seeing it on this page.  I should clarify then that my suggestion would be to make the link to the evaluation center a little more prevalent.

esammann, 2019-04-17:
So I also just realized that the link in the comment below goes to Windows 10 Enterprise.  This page where I am writing this comment is for Windows 10 IoT Enterprise.  I would like to know if there is a trial download for Windows 10 IoT Enterprise.  
Sorry for all of the back to back posts, but I couldn't find a way to just edit the first post.

AndrewBracher, 2019-04-17:
What I have discovered is that Windows 10 IOT Enterprise and Windows 10 Enterprise are actually identical.  Just that IOT is licenced through the LTSC channel.  That leaves the licence key question.  An Enterprise licence is said to be only available through volume licencing. That would mean that once licenced, and the device had shipped, it would then be 'out of contact' with its licence server (KMS or AD) so after 6 months it would fail.  Nice.  Only possibility would seem to be the little advertised OEM licence key channel.  These appear to be just retail keys in effect.  Still to actually get such a key and test this theory out.  Thanks Micro$oft for making this such hard work!  Are you out to punish us?