MicrosoftDocs/windows-iotcore-docs #1121: Support for Arm processors

Eugene0469, 2019-03-04:
Hello. Thank you for the information provided above. Is this procedure applicable to arm processors, given that the minnowboard max uses the 64-bit Intel® Atom™ E38xx Series SoC?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a4a8d9d5-58ae-bbe0-8f71-759f8d40d70f
* Version Independent ID: 8dbf9990-f689-0fb7-70a6-079be6def5cd
* Content: [Mobile Broadband Connection - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/mbb#feedback)
* Content Source: [windows-iotcore/connect-your-device/MBB.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/MBB.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2019-03-05:
Yes.

EmmanuelKinyanjui, 2019-03-06:
Greetings,
I am getting the following issues when I attempt to build the provided project for an ARM target.
Kindly advise on how to rectify this.

1>------ Build started: Project: MBBConnect, Configuration: Debug ARM ------
1>pch.cpp
1>MBBConnect.cpp
1>c:\users\admin\downloads\compressed\iot-utilities-master\iot-utilities-master\mbbconnect\mbbconnect\mbbconnect.cpp(75): warning C4018: '<': signed/unsigned mismatch
1>ARM\Debug\MBBConnect.obj : fatal error LNK1112: module machine type 'x86' conflicts with target machine type 'ARM'
1>Done building project "MBBConnect.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

saraclay, 2019-03-07:
Hi @EmmanuelKinyanjui - [please post this issue on our forums](https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT) as we have staff that is better suited to help you there.

EmmanuelKinyanjui, 2019-03-08:
@saraclay Noted. Thank you!

closed 2019-03-12 by saraclay