MicrosoftDocs/windows-iotcore-docs #1122: Error APPX1101: Payload contains two or more files with the same destination path "Properties\Default.rd.xml"

moyanming, 2019-03-05:
Hi,

Acrroding to the steps [here](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/investigatememoryleaks#create-a-test-app), I got errors after build the solution. I update my VS2017 from v15.9.3 to the v15.9.7, but those errors still happen. How to fix this problem?
[image]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: acf9e618-004c-563d-ad37-e3293a5de5af
* Version Independent ID: 7dabbd00-559b-1a0e-13c9-12cba08f520b
* Content: [Investigating memory leaks - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/investigatememoryleaks#create-a-test-app)
* Content Source: [windows-iotcore/develop-your-app/InvestigateMemoryLeaks.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/InvestigateMemoryLeaks.md)
* Product: **windows-iot**
* GitHub Login: @paulmon
* Microsoft Alias: **paulmon**

saraclay, 2019-03-05:
@moyanming This seems like something our customer support team could help with. Can you file this on our Feedback Hub using the steps [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT)? Thank you.

moyanming, 2019-03-06:
Hi @saraclay ,
Thank you for your reply, I will send this issuse by the Feedback Hub.

saraclay, 2019-03-07:
Thank you @moyanming.

moyanming, 2019-03-08:
The problem solved by remove the Default.rd.xml file from the background project (I jast delete this file and this is the wrong way).
According to the reply [here](https://developercommunity.visualstudio.com/content/problem/477992/error-appx1101-payload-contains-two-or-more-files.html), there should be add one step that is about remove the Default.rd.xml from the background project.

closed 2019-03-11 by saraclay