MicrosoftDocs/windows-iotcore-docs #1123: AllowEmbeddedMode not working any more on Windows 10 Version	10.0.17763 Build 17763

Heris19, 2019-03-05:
Hi,

after updating our dev machines to Windows 10 Enterprise Version 10.0.17763 Build 17763 (1809) we are unable to run and debug UWP background applications.
We get the error "Unable to activate Windows Store app 'xyz'. The activation request failed with error 'This program is blocked by group policy ...'.
I've tried to recreate package with Allow embedded mode but nothing has changed. Seems like the package is not working anymore.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 17c2bf9d-6b31-6748-90ac-68f1f472e3ff
* Version Independent ID: f305b967-06d9-cdc7-5a1c-a699485b9bc7
* Content: [Embedded mode - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/EmbeddedMode)
* Content Source: [windows-iotcore/develop-your-app/EmbeddedMode.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/EmbeddedMode.md)
* Product: **windows-iot**
* GitHub Login: @lilyhou
* Microsoft Alias: **lihou**

gatornation11, 2019-03-05:
It worked for me, no error. Are you running the latest version of the WICD? I had all kinds of issues when I tried to run even a slightly older version of WICD with build 1809.

saraclay, 2019-03-05:
Yes, like @gatornation11, please run the latest version.

Heris19, 2019-03-06:
Yes, I installed WICD yesterday (downloaded from the link on the page - ICD.exe version 10.0.17763.1). Everything run without error, but app won't launch. And we have tried to create the package on multiple machines in our team, but without success. Maybe it could be related to our environment? We have Windows from MSDN, computers connected to Azure AD. Probably all computers are updated to current version of Windows, no new installations of 1809.

Heris19, 2019-03-06:
Maybe could you share us your package? We'd tried it. Thank you.

gatornation11, 2019-03-06:
When you built the package in WICD did you choose "IT admin" as the owner? I attached my package.

[EmbeddedMode.zip](https://github.com/MicrosoftDocs/windows-iotcore-docs/files/2936291/EmbeddedMode.zip)

Heris19, 2019-03-06:
Thank you for the package. Unfortunately nothing has changed.
No, I didn't. Honestly I don't know where to set it up.

closed 2019-04-01 by saraclay