MicrosoftDocs/windows-iotcore-docs #1130: Dragonboard 410c BSP version

endyb66, 2019-03-09:
The last build uses the BSP version 2120.0.0.0, the current one does use the older BSP again?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9fa45d17-9bd5-91d0-feff-676e0ad8fb5a
* Version Independent ID: f9ef63e1-1605-02bf-1d77-b2a07ea0a80b
* Content: [Release Notes for Build 17763.253 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/commercial/17763)
* Content Source: [windows-iotcore/release-notes/commercial/17763.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/commercial/17763.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

saraclay, 2019-03-11:
@endyb66 checking

saraclay, 2019-03-12:
That was a typo on our end - the latest build is 2120.0.0.

closed 2019-03-12 by saraclay