MicrosoftDocs/windows-iotcore-docs #1131: Does it wrong when I set the AllowEmbeddedMode to Yes in the Iot?

moyanming, 2019-03-10:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 17c2bf9d-6b31-6748-90ac-68f1f472e3ff
* Version Independent ID: f305b967-06d9-cdc7-5a1c-a699485b9bc7
* Content: [Embedded mode - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/EmbeddedMode)
* Content Source: [windows-iotcore/develop-your-app/EmbeddedMode.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/EmbeddedMode.md)
* Product: **windows-iot**
* GitHub Login: @lilyhou
* Microsoft Alias: **lihou**

saraclay, 2019-03-11:
Hi @moyanming let me ask a member of our team and will get back to you.

saraclay, 2019-03-11:
Here are the answers:

1. Yes
2. It can't be disabled on IoT Core.
3. It always is.

moyanming, 2019-03-12:
Hi @saraclay ,
Thank you for your answers.

closed 2019-03-12 by moyanming