MicrosoftDocs/windows-iotcore-docs #1133: An error occurred trying to download 'https://iottools.blob.core.windows.net/iotdashboard/Windows10IoTCoreDashboard.application'.

dzhyuliuk, 2019-03-10:
Hi,
It seems iot dashboard setup.exe doesn't work. Something wrong with OneClick app. However, by provided link from error -  app is downloaded and works well. 

Error details:

Log file:
The following properties have been set:
Property: [AdminUser] = true {boolean}
Property: [InstallMode] = HomeSite {string}
Property: [NTProductType] = 1 {int}
Property: [ProcessorArchitecture] = AMD64 {string}
Property: [VersionNT] = 10.0.0 {version}
Launching Application.
URLDownloadToCacheFile failed with HRESULT '-2146697208'
Error: An error occurred trying to download 'https://iottools.blob.core.windows.net/iotdashboard/Windows10IoTCoreDashboard.application'.

saraclay, 2019-03-11:
Thank you for letting us know @dzhyuliuk. Do you mind filing this with our Feedback Hub? That way it's triaged? I'll also give a heads-up to the team. Thanks!

Instructions for Feedback Hub are [here](https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT).

saraclay, 2019-03-11:
@dzhyuliuk, I checked with our team and currently this is not a known issue. Once you file this in Feedback Hub, could you provide me with the title of the issue? Thanks!

closed 2019-04-01 by saraclay