MicrosoftDocs/windows-iotcore-docs #1145: The links of the samples are all lose

moyanming, 2019-03-13:
Hi,
The [links of the samples](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/lightningproviders#lightning-provider-samples) are all lose.

BTW, does the lightning provider is not support at all?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e7151dc6-2eb0-5587-af76-27a9df08a9d2
* Version Independent ID: 97d75f37-053d-3d91-e6e5-6df60a4bb690
* Content: [Lightning providers - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/lightningproviders)
* Content Source: [windows-iotcore/develop-your-app/LightningProviders.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/LightningProviders.md)
* Product: **windows-iot**
* GitHub Login: @msalehmsft
* Microsoft Alias: **msaleh**

saraclay, 2019-03-18:
@moyanming I apologize for the lack of response from me. Yes - thank you for pointing that out. Lightning is not actively being supported, but it would help to know in what capacity you're using it for.

saraclay, 2019-03-19:
@moyanming, thank you for your patience.

If you're using our latest build, lightning should work. I've also updated the sample links, sorry about that! Please note that we tried this on a RPi3. If you're finding that this is not working, feel free to follow up on this thread.

closed 2019-04-01 by saraclay